libimobiledevice/libirecovery #13: problem when run make with fedora v 21

mamislimen, 2015-01-24:
when i run ./aurogen.sh is work perfect without error but when i run make he don't make anything
help me please 
[root@localhost libirecovery]# ./autogen.sh 
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... no
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... no
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libusb... yes
checking for ANSI C header files... (cached) yes
checking for stdint.h... (cached) yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking for an ANSI C-conforming const... yes
checking for size_t... yes
checking for ssize_t... yes
checking for uint16_t... yes
checking for uint32_t... yes
checking for uint8_t... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible malloc... yes
checking for stdlib.h... (cached) yes
checking for GNU libc compatible realloc... yes
checking for strcasecmp... yes
checking for strdup... yes
checking for strerror... yes
checking for strndup... yes
checking whether to enable WIN32 build settings... no
checking for supported compiler flags...  -Wall -Wextra -Wmissing-declarations -Wredundant-decls -Wshadow -Wpointer-arith -Wwrite-strings -Wswitch-default -Wno-unused-parameter -fvisibility=hidden
checking whether make supports nested variables... (cached) yes
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating include/Makefile
config.status: creating tools/Makefile
config.status: creating libirecovery.pc
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

Configuration for libirecovery 0.2.0:
-------------------------------------------

  Install prefix: .........: /usr/local

  Now type 'make' to build libirecovery 0.2.0,
  and then 'make install' for installation.

[root@localhost libirecovery]# make
make  all-recursive
make[1]: Entering directory '/root/Téléchargements/libirecovery'
Making all in src
make[2]: Entering directory '/root/Téléchargements/libirecovery/src'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/Téléchargements/libirecovery/src'
Making all in include
make[2]: Entering directory '/root/Téléchargements/libirecovery/include'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/Téléchargements/libirecovery/include'
Making all in tools
make[2]: Entering directory '/root/Téléchargements/libirecovery/tools'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/Téléchargements/libirecovery/tools'
make[2]: Entering directory '/root/Téléchargements/libirecovery'
make[2]: Leaving directory '/root/Téléchargements/libirecovery'
make[1]: Leaving directory '/root/Téléchargements/libirecovery'
[root@localhost libirecovery]#

nikias, 2015-01-24:
The output looks like it was already built. If you now type ```make clean``` followed by ```make``` do you see it compiling?

mamislimen, 2015-01-25:
@nikias thanks for you answer but when i try same error

[root@localhost libirecovery]# make clean
Making clean in src
make[1]: Entering directory '/root/Téléchargements/libirecovery/src'
test -z "libirecovery.la" || rm -f libirecovery.la
rm -f ./so_locations
rm -rf .libs _libs
rm -f *.o
rm -f *.lo
make[1]: Leaving directory '/root/Téléchargements/libirecovery/src'
Making clean in include
make[1]: Entering directory '/root/Téléchargements/libirecovery/include'
rm -rf .libs _libs
rm -f *.lo
make[1]: Leaving directory '/root/Téléchargements/libirecovery/include'
Making clean in tools
make[1]: Entering directory '/root/Téléchargements/libirecovery/tools'
 rm -f irecovery
rm -rf .libs _libs
rm -f *.o
rm -f *.lo
make[1]: Leaving directory '/root/Téléchargements/libirecovery/tools'
make[1]: Entering directory '/root/Téléchargements/libirecovery'
rm -rf .libs _libs
rm -f *.lo
make[1]: Leaving directory '/root/Téléchargements/libirecovery'
[root@localhost libirecovery]# make 
make  all-recursive
make[1]: Entering directory '/root/Téléchargements/libirecovery'
Making all in src
make[2]: Entering directory '/root/Téléchargements/libirecovery/src'
  CC       libirecovery_la-libirecovery.lo
  CCLD     libirecovery.la
make[2]: Leaving directory '/root/Téléchargements/libirecovery/src'
Making all in include
make[2]: Entering directory '/root/Téléchargements/libirecovery/include'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/root/Téléchargements/libirecovery/include'
Making all in tools
make[2]: Entering directory '/root/Téléchargements/libirecovery/tools'
  CC       irecovery-irecovery.o
  CCLD     irecovery
make[2]: Leaving directory '/root/Téléchargements/libirecovery/tools'
make[2]: Entering directory '/root/Téléchargements/libirecovery'
make[2]: Leaving directory '/root/Téléchargements/libirecovery'
make[1]: Leaving directory '/root/Téléchargements/libirecovery'
[root@localhost libirecovery]# sudo make install
Making install in src
make[1]: Entering directory '/root/Téléchargements/libirecovery/src'
make[2]: Entering directory '/root/Téléchargements/libirecovery/src'
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/sh ../libtool   --mode=install /usr/bin/install -c   libirecovery.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libirecovery.so.2.0.1 /usr/local/lib/libirecovery.so.2.0.1
libtool: install: (cd /usr/local/lib && { ln -s -f libirecovery.so.2.0.1 libirecovery.so.2 || { rm -f libirecovery.so.2 && ln -s libirecovery.so.2.0.1 libirecovery.so.2; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libirecovery.so.2.0.1 libirecovery.so || { rm -f libirecovery.so && ln -s libirecovery.so.2.0.1 libirecovery.so; }; })
libtool: install: /usr/bin/install -c .libs/libirecovery.lai /usr/local/lib/libirecovery.la
libtool: install: /usr/bin/install -c .libs/libirecovery.a /usr/local/lib/libirecovery.a
libtool: install: chmod 644 /usr/local/lib/libirecovery.a
libtool: install: ranlib /usr/local/lib/libirecovery.a
libtool: finish: PATH="/sbin:/bin:/usr/sbin:/usr/bin:/sbin" ldconfig -n /usr/local/lib
----------------------------------------------------------------------
Libraries have been installed in:
   /usr/local/lib

If you ever happen to want to link against installed libraries
in a given directory, LIBDIR, you must either use libtool, and
specify the full pathname of the library, or use the `-LLIBDIR'
flag during linking and do at least one of the following:
   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable
[... 19 lines cut ...]
make[1]: Leaving directory '/root/Téléchargements/libirecovery/include'
Making install in tools
make[1]: Entering directory '/root/Téléchargements/libirecovery/tools'
make[2]: Entering directory '/root/Téléchargements/libirecovery/tools'
 /usr/bin/mkdir -p '/usr/local/bin'
  /bin/sh ../libtool   --mode=install /usr/bin/install -c irecovery '/usr/local/bin'
libtool: install: /usr/bin/install -c .libs/irecovery /usr/local/bin/irecovery
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/root/Téléchargements/libirecovery/tools'
make[1]: Leaving directory '/root/Téléchargements/libirecovery/tools'
make[1]: Entering directory '/root/Téléchargements/libirecovery'
make[2]: Entering directory '/root/Téléchargements/libirecovery'
make[2]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/usr/local/lib/pkgconfig'
 /usr/bin/install -c -m 644 libirecovery.pc '/usr/local/lib/pkgconfig'
make[2]: Leaving directory '/root/Téléchargements/libirecovery'
make[1]: Leaving directory '/root/Téléchargements/libirecovery'
[root@localhost libirecovery]# ls
aclocal.m4      config.h.in    COPYING          libirecovery.pc.in  Makefile.in
autogen.sh      config.log     depcomp          libtool             missing
autom4te.cache  config.status  idevicerestore   ltmain.sh           README
compile         config.sub     include          m4                  src
config.guess    configure      install-sh       Makefile            stamp-h1
config.h        configure.ac   libirecovery.pc  Makefile.am         tools
[root@localhost libirecovery]#

nikias, 2015-01-25:
It compiled! See:
```
Making all in src
make[2]: Entering directory '/root/Téléchargements/libirecovery/src'
CC libirecovery_la-libirecovery.lo
CCLD libirecovery.la
```
and you also installed it, so you can try to run the ```irecovery``` command now.

mamislimen, 2015-01-25:
@nikias  thank you very much for your help
it work now :+1:

mamislimen, 2015-01-25:
@nikias  an other problem when i compile  idevicerestore he say "No package 'libirecovery' found"
 please help
[root@localhost idevicerestore]# ./autogen.sh 
libtoolize: putting auxiliary files in `.'.
libtoolize: linking file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: linking file `m4/libtool.m4'
libtoolize: linking file `m4/ltoptions.m4'
libtoolize: linking file `m4/ltsugar.m4'
libtoolize: linking file `m4/ltversion.m4'
libtoolize: linking file `m4/lt~obsolete.m4'
libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
configure.ac:15: installing './compile'
checking whether make supports nested variables... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/root/idevicerestore/missing: Unknown `--is-lightweight' option
Try `/root/idevicerestore/missing --help' for more information
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert x86_64-unknown-linux-gnu file names to x86_64-unknown-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-unknown-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... no
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for libirecovery... no
configure: error: Package requirements (libirecovery >= 0.2.0) were not met:

No package 'libirecovery' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libirecovery_CFLAGS
and libirecovery_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
[root@localhost idevicerestore]#

mamislimen, 2015-01-30:
@nikias please help me

FunkyM, 2015-09-25:
The initially reported problem was solved.
Please use the "idevicerestore" issues for questions regarding that.
Your new problem probably arises from the fact that you installed in "/usr/local".
Try to use "--prefix=/usr" instead for all configure steps or even with "./autogen.sh --prefix=/usr".

closed 2015-09-25 by FunkyM

mamislimen, 2015-09-25:
@FunkyM  thank you very much for your help

kristenlc, 2022-09-09:
It is now September 8, 2022. I have gone through 6 hours of headaches trying to get idevicerestore to compile, trying every possible source on the internet. All leads back to this error message. macOS High Sierra 10.13.6, Xcode 9.4.1.

I finally got it to work after compiling and installing libirecovery from source, and creating symlinks manually by hand.

until now I have tried various different versions of irecovery, I have tried adding -prefix=/usr to ./autogen.sh before compiling and I have tried without that. I have tried installing from the homebrew formula, I have tried linking with homebrew after installing from source, I have tried installing libirecovery from homebrew then removing the files in /usr/local/lib for irecovery and installing in the default prefix from source and then relinking using the command line tools with homebrew. I have tried macports install of libirecovery. I have tried every single possible thing mentioned on the internet that I could find in 6 hours, and no luck.

I am going to attach what fixed this error for me and I hope it benefits someone, or two, or a hundred people.

> cd /usr/lib
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.0.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.0.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.0.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery.la 
> sudo ln -s libirecovery-1.0.a libirecovery.a
> sudo ln -s libirecovery-1.0.dylib libirecovery.dylib 
> 
> sudo ln -s libplist-2.0.la libplist.la
> sudo ln -s libplist-2.0.a libplist.a
> sudo ln -s libplist-2.0.dylib libplist.dylib
> 
> sudo ln -s libplist++-2.0.la libplist++.la # maybe not necessary 
> sudo ln -s libplist++-2.0.a libplist++.a  # maybe not necessary 
> sudo ln -s libplist++-2.0.dylib libplist++.dylib # maybe not necessary 
> 
> cd pkgconfig
> 
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.0.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery.pc
> 
> sudo ln -s libplist-2.0.pc libplist.pc
> sudo ln -s libplist++-2.0.pc libplist++.pc # maybe not necessary
> 
> cd /usr/local/lib
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery-0.2.0.la # maybe not necessary 
> sudo ln -s libirecovery-1.0.a libirecovery-0.2.0.a # maybe not necessary 
> sudo ln -s libirecovery-1.0.dylib libirecovery-0.2.0.dylib # maybe not necessary 
> 
> sudo ln -s libirecovery-1.0.la libirecovery.la 
> sudo ln -s libirecovery-1.0.a libirecovery.a
> sudo ln -s libirecovery-1.0.dylib libirecovery.dylib 
> 
> sudo ln -s libplist-2.0.la libplist.la
> sudo ln -s libplist-2.0.a libplist.a
> sudo ln -s libplist-2.0.dylib libplist.dylib
> 
> sudo ln -s libplist++-2.0.la libplist++.la # maybe not necessary 
> sudo ln -s libplist++-2.0.a libplist++.a  # maybe not necessary 
> sudo ln -s libplist++-2.0.dylib libplist++.dylib # maybe not necessary 
> 
> cd pkgconfig
> 
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery-0.2.0.pc # maybe not necessary
> sudo ln -s libirecovery-1.0.pc libirecovery.pc
> 
> sudo ln -s libplist-2.0.pc libplist.pc
> sudo ln -s libplist++-2.0.pc libplist++.pc # maybe not necessary

Explanation

It is looking for a package named libirecovery but the installed one on the system is libirecovery-1.0, so it is saying no package found with that name. Ok, creating a symlink so libirecovery is an alias for libirecovery-1.0 on the system, fixes that issue. Simple enough right? Well I wish the internet would Have let me know about that 6 hours ago. The same thing goes for libplist, which must be installed the latest version from source and not from homebrew as the formula on homebrew is outdated. That is all, k thx, bye.

kristenlc, 2022-09-09:
If you get this error when running `make` after the above steps

```
/usr/local/include/libgeneral/macros.h:55:13:` error: expected "FILENAME" or
      <FILENAME>
#   include CUSTOM_LOGGING
```

You can fix by 

> git clone --recursive https://github.com/tihmstar/libgeneral.git
> cd libgeneral
> git checkout 636786a13dd16dc75bbfd1c20557095e9a1c5bce
> ./autogen.sh
> sudo rm -rf /usr/local/lib/libgeneral*
> sudo rm -rf /usr/local/lib/pkgconfig/libgeneral*
> sudo rm -rf /usr/local/include/libgeneral*
> make clean
> make
> sudo make install
> ls /usr/local/lib/libgeneral*

But there is nothing you can do to fix the ____chkstk_darwin symbol(s) not found you get on high sierra 10.13
```
  "____chkstk_darwin", referenced from:
      _decompress_lzss in libxpwn.a(lzss.c.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [futurerestore] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```