libimobiledevice/libirecovery #15: configure: error: Package requirements (libusb-1.0 >= 1.0.3) were not met:  No package 'libusb-1.0' found

mcjoeng, 2015-03-09:
[image]

Help ! 
I have installed libusb-1.0 from http://download.opensuse.org/repositories/home:/FunkyM:/iOS/openSUSE_13.2/ . 
But it noticed me that "configure: error: Package requirements (libusb-1.0 >= 1.0.3) were not met:  No package 'libusb-1.0' found"

mcjoeng, 2015-03-09:
Is it the problem of libusb-0_1-4?

[image]

mcjoeng, 2015-03-09:
[image]

I find two libusb installed

nikias, 2015-03-09:
You need to install ```libusb-1_0-devel```. This is a default procedure and not an issue with libirecovery.

closed 2015-03-09 by nikias