libimobiledevice/libirecovery #22: Release for homebrew integration

ebekebe, 2016-05-18:
libirecovery is a requirement of idevicerestore, which expects at least version 0.0.2. I currently try to make a homebrew formula that builds idevicerestore and for that I created a formula for libirecovery. Since homebrew accepts only tagged versions it would be cool if you could make a new 0.0.2 release, so I can progress with the formula.

Related pull request: https://github.com/Homebrew/homebrew-core/pull/1259

awesomebing1, 2016-06-07:
I would like to see this as well, please.

bjoernhaeuser, 2016-06-10:
That would be really amazing :)

spotlightishere, 2017-07-03:
Is there any update about having a release for this?

FunkyM, 2020-06-15:
Closing as version [1.0.0](/libimobiledevice/libirecovery/releases/tag/1.0.0) has been released.

closed 2020-06-15 by FunkyM