libimobiledevice/libirecovery #24: irecovery not properly grabbing nonce

awesomebing1, 2016-08-08:
I found an interesting problem that I think might be a hardware issue (on the iPhone, not the computer) but would like to put it here so other people can examine. I'm using Ubuntu 16.10.

I was messing around with an iPhone 3GS that has a bricked baseband or something like that, and was trying to downgrade it to iOS 3.0 using `idevicerestore`. I hadn't booted it into pwned dfu or anything like that. I noticed that it kept failing, with similar output each time:

```
Reading data from version.xml
NOTE: using cached version data
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Found device in Recovery mode
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Identified device as n88ap, iPhone2,1
Extracting Restore.plist from IPSW
Product Version: 3.0
Product Build: 7A341 Major: 7
opening device 05ac:1281...
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
Setting to configuration 1
Setting to interface 0:0
Device supports Image4: false
Custom firmware requested. Disabled TSS request.
Variant: Customer Erase Install (IPSW)
This restore will erase your device data.
Using cached filesystem from '/home/me/Projects/iOS/Bundles/iPhone2,1_3.0_7A341_Restore/018-5302-002.dmg'
Extracting iBEC.n88ap.RELEASE.dfu...
Not personalizing component iBEC...
Writing data to iBEC
Sending iBEC (104772 bytes)...
ERROR: could not retrieve device serial number. Can't continue.
```

I was kind of confused, so I went into irecovery (where irecv_copy_nonce_with_tag is) and using my basic C skills, I added `debug("Magically, a tag appeared: %s\n", nonce);` right below [src/libirecovery.c#L332](https://github.com/libimobiledevice/libirecovery/blob/686d6d72d56e700c13abbb75a31ecb385658d96d/src/libirecovery.c#L332). 

This changed the output to the following (which is a snippet):
```
Magically, a tag appeared: 
irecv_copy_nonce_with_tag: WARNING: couldn't find tag NONC in string Apple Inc.
Magically, a tag appeared: 
irecv_copy_nonce_with_tag: WARNING: couldn't find tag SNON in string Apple Inc.
```
The tag passed to `irecv_copy_nonce_with_tag` appears to be null, if I understood printing stuff in C correctly. So, I went up to [src/libirecovery.c#L650](https://github.com/libimobiledevice/libirecovery/blob/686d6d72d56e700c13abbb75a31ecb385658d96d/src/libirecovery.c#L650) and added the following:
```
debug("Your ap: %s\n", &client->device_info.ap_nonce);
debug("Your sep: %s\n", &client->device_info.sep_nonce);
debug("Here's a bonus for working with us: %s\n", serial_str);
```

Well, this produced the following snippet:
```
Your ap: 
Your sep: 
Here's a bonus for working with us: CPID:8920 CPRV:14 CPFM:03 SCEP:03 BDID:00 ECID:redacted IBFL:01 IMEI:[redacted]
```

ap and sep are both empty as well. I honestly don't know what's wrong here, perhaps it's hardware, libirecovery, or my C skills aren't outputting anything because I messed up.

(If you think this is an idevicerestore issue, please tell me.)

awesomebing1, 2016-08-09:
I was able to reproduce the same thing (without debugging) on the same Mac but using 10.11.6.

spotlightishere, 2019-04-18:
Turns out if you accidentally overwrite the NOR and are too technologically inexperienced, it will not send one as expected. iTunes restores fine with this but this tool does not. Upon rewriting the NOR for this 3GS it was a-okay.

closed 2019-04-18 by spotlightishere

Blefish, 2019-04-18:
Maybe this could be implemented into imobiledevice?