libimobiledevice/libirecovery #28: Q: Use irecovery/irestore to ssh/mount root (/)?

alexli09, 2017-02-09:
Greetings,

Is there any way to use your tools to achieve the above mentioned? I'm trying to rescue some photos before wiping out the phone.

My phone won't boot now. I believe the culprit is a full root (/). The Delete function in Photos stopped working.

I tried everything else, like make_noerase_ipsw, and ran into different roadblocks.

My phone:
 - iPhone 5s
 - Probably on iOS 8.x

Or, if there's a way for me to  free up some space in root (/) and try reboot, or just copy those photos out before I do a full restore.

Or, using your tools to perform an equivalent of "Update". Tried that in iTunes and it would stop dead with an Apple logo and one tiny blob on the progress bar and eventually timed out.

Also the interactive shell from "irecovery -s" leads me to think that it may have the ability to deal with the filesystems except there are no documentations I can find regarding the commands available. The only example that kept coming up in Google is the "setenv/saveenv" example.

Any pointers or tricks to try are much appreciated. My phone is not jailbroken.

Cheers,

Alex

alexli09, 2017-02-10:
Greetings,

Some questions:

irecovery -s: what does this shell interface with?

I came across some commands on the web and thought just try them.Like:

`nvram auto-boot=true
nvram -d boot-args
fsck_hfs -fy /dev/disk0s1
/sbin/mount_hfs -o noasync,sync /dev/disk0s1 /mnt1
/sbin/mount_hfs -o noasync,sync /dev/disk0s2 /tmp
mv /tmp/mobile/Library/Preferences /tmp/Prefs.mobile
mv /tmp/root/Library/Preferences /tmp/Prefs.root
echo "RELOACATED PREFERENCES"
sleep 30
reboot`

Sorry the line breaks don't seem to work.

Nothing tells me it works or not except 'reboot' which kicks me out of the shell.

There's no stdout or stderr, or this shell is not suppose to show anything.

Really just trying to get my phone to boot normally or remove a few large files so the root (/) isn't so full and the ability to copy some photos out of it.

Cheers!

Alex

nikias, 2017-03-13:
This is not possible.

closed 2017-03-13 by nikias