libimobiledevice/libirecovery #32: Trying to build on msys2

rcmpayne, 2017-06-07:
getting error on make while building on msys2

```
config.status: creating Makefile
config.status: creating src/Makefile
config.status: creating include/Makefile
config.status: creating tools/Makefile
config.status: creating libirecovery.pc
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands

Configuration for libirecovery 0.2.0:
-------------------------------------------

  Install prefix: .........: /usr
  USB backend: ............: libusb 1.0.21

  Now type 'make' to build libirecovery 0.2.0,
  and then 'make install' for installation.
```
```
# make
make  all-recursive
make[1]: Entering directory '/home/rpayne/github/libirecovery-master'
Making all in src
make[2]: Entering directory '/home/rpayne/github/libirecovery-master/src'
  CC       libirecovery_la-libirecovery.lo
libirecovery.c: In function ‘mobiledevice_connect’:
libirecovery.c:505:2: error: unknown type name ‘SP_DEVICE_INTERFACE_DATA’
  SP_DEVICE_INTERFACE_DATA currentInterface;
  ^~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:506:2: error: unknown type name ‘HDEVINFO’
  HDEVINFO usbDevices;
  ^~~~~~~~
libirecovery.c:512:15: warning: implicit declaration of function ‘SetupDiGetClassDevs’ [-Wimplicit-function-declaration]
  usbDevices = SetupDiGetClassDevs(&GUID_DEVINTERFACE_DFU, NULL, NULL, DIGCF_PRESENT | DIGCF_DEVICEINTERFACE);
               ^~~~~~~~~~~~~~~~~~~
libirecovery.c:512:71: error: ‘DIGCF_PRESENT’ undeclared (first use in this function)
  usbDevices = SetupDiGetClassDevs(&GUID_DEVINTERFACE_DFU, NULL, NULL, DIGCF_PRESENT | DIGCF_DEVICEINTERFACE);
                                                                       ^~~~~~~~~~~~~
libirecovery.c:512:71: note: each undeclared identifier is reported only once for each function it appears in
libirecovery.c:512:87: error: ‘DIGCF_DEVICEINTERFACE’ undeclared (first use in this function)
  usbDevices = SetupDiGetClassDevs(&GUID_DEVINTERFACE_DFU, NULL, NULL, DIGCF_PRESENT | DIGCF_DEVICEINTERFACE);
                                                                                       ^~~~~~~~~~~~~~~~~~~~~
libirecovery.c:513:41: error: ‘SP_DEVICE_INTERFACE_DATA’ undeclared (first use in this function)
  memset(&currentInterface, '\0', sizeof(SP_DEVICE_INTERFACE_DATA));
                                         ^~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:514:18: error: request for member ‘cbSize’ in something not a structure or union
  currentInterface.cbSize = sizeof(SP_DEVICE_INTERFACE_DATA);
                  ^
libirecovery.c:515:27: warning: implicit declaration of function ‘SetupDiEnumDeviceInterfaces’ [-Wimplicit-function-declaratio ]
  for(i = 0; usbDevices && SetupDiEnumDeviceInterfaces(usbDevices, NULL, &GUID_DEVINTERFACE_DFU, i, &currentInterface); i++) {
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:522:3: error: unknown type name ‘PSP_DEVICE_INTERFACE_DETAIL_DATA’
   PSP_DEVICE_INTERFACE_DETAIL_DATA details;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:523:3: warning: implicit declaration of function ‘SetupDiGetDeviceInterfaceDetail’ [-Wimplicit-function-declaration]
   SetupDiGetDeviceInterfaceDetail(usbDevices, &currentInterface, NULL, 0, &requiredSize, NULL);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:524:14: error: ‘PSP_DEVICE_INTERFACE_DETAIL_DATA’ undeclared (first use in this function)
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:524:48: error: expected ‘;’ before ‘malloc’
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
                                                ^~~~~~
libirecovery.c:525:10: error: invalid type argument of ‘->’ (have ‘int’)
   details->cbSize = sizeof(SP_DEVICE_INTERFACE_DETAIL_DATA);
          ^~
libirecovery.c:525:28: error: ‘SP_DEVICE_INTERFACE_DETAIL_DATA’ undeclared (first use in this function)
   details->cbSize = sizeof(SP_DEVICE_INTERFACE_DETAIL_DATA);
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:527:9: warning: passing argument 1 of ‘free’ makes pointer from integer without a cast [-Wint-conversion]
    free(details);
         ^~~~~~~
In file included from /usr/include/stdio.h:29:0,
                 from libirecovery.c:25:
/usr/include/stdlib.h:93:7: note: expected ‘void *’ but argument is of type ‘int’
 _VOID _EXFUN_NOTHROW(free,(_PTR));
       ^
libirecovery.c:531:34: error: invalid type argument of ‘->’ (have ‘int’)
    memcpy((void*) result, details->DevicePath, requiredSize - sizeof(DWORD));
                                  ^~
libirecovery.c:532:9: warning: passing argument 1 of ‘free’ makes pointer from integer without a cast [-Wint-conversion]
    free(details);
         ^~~~~~~
In file included from /usr/include/stdio.h:29:0,
                 from libirecovery.c:25:
/usr/include/stdlib.h:93:7: note: expected ‘void *’ but argument is of type ‘int’
 _VOID _EXFUN_NOTHROW(free,(_PTR));
       ^
libirecovery.c:591:2: warning: implicit declaration of function ‘SetupDiDestroyDeviceInfoList’ [-Wimplicit-function-declaratio ]
  SetupDiDestroyDeviceInfoList(usbDevices);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:601:18: error: request for member ‘cbSize’ in something not a structure or union
  currentInterface.cbSize = sizeof(SP_DEVICE_INTERFACE_DATA);
                  ^
libirecovery.c:609:36: error: expected ‘;’ before ‘details’
   PSP_DEVICE_INTERFACE_DETAIL_DATA details;
                                    ^~~~~~~
libirecovery.c:611:3: error: ‘details’ undeclared (first use in this function)
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
   ^~~~~~~
libirecovery.c:611:48: error: expected ‘;’ before ‘malloc’
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
                                                ^~~~~~
make[2]: *** [Makefile:455: libirecovery_la-libirecovery.lo] Error 1
make[2]: Leaving directory '/home/rpayne/github/libirecovery-master/src'
make[1]: *** [Makefile:466: all-recursive] Error 1
make[1]: Leaving directory '/home/rpayne/github/libirecovery-master'
make: *** [Makefile:375: all] Error 2
```

Orbif, 2018-01-03:
Try adding LIBS="-lws2_32" when running the autogen.sh

**./autogen.sh LIBS="-lws2_32"**

vege226, 2018-02-02:
Have the same problem.

./autogen.sh LIBS="-lws2_32"

 did not work for me!

vege226, 2018-02-02:
I add 
/* AGREGADAS */
typedef PVOID HDEVINFO;

#define DIGCF_DEFAULT	0x00000001
#define DIGCF_PRESENT	0x00000002
#define DIGCF_DEVICEINTERFACE	0x00000010

typedef struct _SP_DEVICE_INTERFACE_DETAIL_DATA {
  DWORD cbSize;
  TCHAR DevicePath[ANYSIZE_ARRAY];
} SP_DEVICE_INTERFACE_DETAIL_DATA, *PSP_DEVICE_INTERFACE_DETAIL_DATA;

typedef struct _SP_DEVICE_INTERFACE_DATA {
    DWORD cbSize;
    GUID  InterfaceClassGuid;
    DWORD Flags;
    ULONG_PTR Reserved;
} SP_DEVICE_INTERFACE_DATA, *PSP_DEVICE_INTERFACE_DATA;

/* AGREGADAS */

in libirecovery.c

seems working

rcmpayne, 2018-03-07:
yea still fails

`rcmpayne@dev MSYS ~/github/libirecovery-master/libirecovery-master
# ./autogen.sh LIBS="-lws2_32"
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:26: installing './compile'
configure.ac:6: installing './missing'
src/Makefile.am: installing './depcomp'
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether make supports nested variables... (cached) yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking build system type... x86_64-pc-msys
checking host system type... x86_64-pc-msys
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/x86_64-pc-msys/bin/ld.exe
checking if the linker (/usr/x86_64-pc-msys/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-pc-msys file names to x86_64-pc-msys format... func_convert_file_noop
checking how to convert x86_64-pc-msys file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-msys/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... gcc -E
[... 157 lines cut ...]
                 from libirecovery.c:25:
/usr/include/stdlib.h:93:7: note: expected ‘void *’ but argument is of type ‘int’
 _VOID _EXFUN_NOTHROW(free,(_PTR));
       ^
libirecovery.c:597:2: warning: implicit declaration of function ‘SetupDiDestroyDeviceInfoList’ [-Wimplicit-function-declaration]
  SetupDiDestroyDeviceInfoList(usbDevices);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
libirecovery.c:607:18: error: request for member ‘cbSize’ in something not a structure or union
  currentInterface.cbSize = sizeof(SP_DEVICE_INTERFACE_DATA);
                  ^
libirecovery.c:615:36: error: expected ‘;’ before ‘details’
   PSP_DEVICE_INTERFACE_DETAIL_DATA details;
                                    ^~~~~~~
libirecovery.c:617:3: error: ‘details’ undeclared (first use in this function)
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
   ^~~~~~~
libirecovery.c:617:48: error: expected ‘;’ before ‘malloc’
   details = (PSP_DEVICE_INTERFACE_DETAIL_DATA) malloc(requiredSize);
                                                ^~~~~~
make[2]: *** [Makefile:457: libirecovery_la-libirecovery.lo] Error 1
make[2]: Leaving directory '/home/rcmpayne/github/libirecovery-master/libirecovery-master/src'
make[1]: *** [Makefile:468: all-recursive] Error 1
make[1]: Leaving directory '/home/rcmpayne/github/libirecovery-master/libirecovery-master'
make: *** [Makefile:377: all] Error 2
`

rcmpayne, 2018-03-28:
Can anyone help heres? i've tried msys, msys2, 32b, x64 and i cant build libirecovery which is needed to build idevicerestore.

okhodjaev-aurea, 2018-04-22:
I've just encountered the same issue on Cygwin

0make, 2020-05-20:
maybe you can try to add -lSetupapi, I resolved a similar issue in another project

Merculous, 2022-06-30:
I have the same issue here on the current commit. I'm unable to compile, getting the same error as original author.