libimobiledevice/libirecovery #33: iPod Touch1'2'3  the BM of  firmwares ,  CPID  is wrong?

4kzeng, 2017-06-22:
A question, see below:
libirecovery.cpp:
	{"iPod2,1",    "n72ap", 0x00, 0x8920 },
	{"iPod3,1",    "n18ap", 0x02, 0x8922 },

BM of firmware:
		<dict>
			<key>ApBoardID</key>
			<string>0x00</string>
			<key>ApChipID</key>
			<string>0x8720</string>
Is there '8920' should change to  '8720'? 
But no device to prove that.

nikias, 2017-06-25:
I also don't have a device, and the BuildManifest indeed contains a value of 0x8720... However this commit was probably not done without a reason @FunkyM? https://github.com/libimobiledevice/libirecovery/commit/3687ea11271b79fb2643c8f041490544c17c5e58

nikias, 2018-01-03:
This was fixed with #36, tested with an actual device.

closed 2018-01-03 by nikias