libimobiledevice/libirecovery #40: iPad6,11 - J72tAP: [Error] ERROR: Unable to discover device model [Error] Fail code=-2

cartwright118, 2018-08-07:
Hi,

Could someone please assist. Having an issue trying to restore to 11.2.6. Using Future Restore on MacOS, most recent version and dependencies.

Using an iPad6,11 - J72tAP with (assumed) valid SHSH blobs from TSSSAVER. I've seen a post here > encounter#32 and checked libirecovery.c and can see: {"iPad6,11", "j71tap", 0x10, 0x8003 } is present

I've also referred to exact file paths rather than relative during command execution

Please see below for paste of terminal

Could someone please advise?

Thanks,
Christian

Version: b99eb81 - 152
Libipatcher Version: f32e41d850f51448bd6c588ead9c7d6455733f3c - 44
Odysseus Support: yes
[INFO] 64bit device detected
futurerestore init done
reading ticket /Users/user/Downloads/downgrade/blob.shsh2 done
[Error] ERROR: Unable to discover device model
[Error] Fail code=-2
Failed with errorcode=-2

Blefish, 2018-08-08:
So you have an iPad6,11 with j72tap? This doesn't seem to be included in commit https://github.com/libimobiledevice/libirecovery/commit/260cb0684bf1cb7d2cdef8ce4a1ec6d832eb1d1c

nikias, 2018-08-10:
@cartwright118 can you run `ideviceinfo -k HardwareModel` and `ideviceinfo -k ProductType`?

cartwright118, 2018-08-13:
Hi guys,

Sorry, just noticed that I made a mistake with copy and paste from libirecovery.c when making my post. My device is:  iPad6,12 - J72tAP as confirmed with ideviceinfo below:

$ ideviceinfo -k HardwareModel
J72tAP
$ ideviceinfo -k ProductType
iPad6,12

When I first started trying to future restore, this value was not present in libirecovery.c included in future restore, so I re-cloned and re-installed from /libimobiledevice/libirecovery/ which included my model but the error is still showing in Future Restore:

[Error] ERROR: Unable to discover device model
[Error] Fail code=-2

As ideviceinfo is able to show my correct model, would you say that the issue is more so with Future Restore?   I did also see some reference online to iPad 5 blobs could potentially be invalid due to multiple board configs, could that be related in any way?

Thanks,
Christian

nikias, 2018-08-13:
ideviceinfo is just querying the device for information. futurerestore is a different project so you need to seek help there. This is not a bug in libirecovery.

closed 2018-08-13 by nikias