libimobiledevice/libirecovery #41: WIN32: Libirecovery is not working with Apple device drivers 423.36.0.0 from Windows Update

taanders, 2018-08-28:
libirecovery is not working with Apple device drivers 423.36.0.0 from Window Update. I tried to use idevicerestore -e -l command with iPhone 5S model and it cannot complete. iTunes can do restore process with this Windows Update driver and older Apple delivered driver from AppleApplicationSupport.msi.

Not working driver:
'''
Original Name:      appleusb.inf
Provider Name:      Apple, Inc.
Class Name:         Universal Serial Bus -laitteet
Class GUID:         {88bae032-5a81-49f0-bc3d-a4ff138216d6}
Driver Version:     05/07/2018 423.36.0.0
Signer Name:        Microsoft Windows Hardware Compatibility Publisher
'''

Device instance path line seems to be changed to line like on libirecovery mobiledevice_connect(recv_client_t* client, unsigned long long ecid) "get iBoot path" line 669 :
'''
\\?\usb#vid_05ac&pid_1281&mi_00#8&37932c9b&0&0000#{ed82a167-d61a-4af6-9ab6-11e52236c576}\usb#vid_05ac&pid_1281#cpid:8960_cprv:11_cpfm:03_scep:01_bdid:02_ecid:000007f6f24358d0_ibfl:1d_srnm:[dnplr9abffgc]#mux1
'''

I modified that code to like this and was able to move little longer on idevicerestore process with those drivers.

My modified code:

'''
            char serial_str[256];
	    serial_str[0] = '\0';
            if (((sscanf(result, "\\\\?\\usb#vid_%*04x&pid_%*04x#%s#", serial_str) != 1) && (sscanf(result, "\\\\?\\usb#%s\\#vid_%*04x&pid_%*04x#%s#", serial_str) != 1)) || (serial_str[0] == '\0')) {
				mobiledevice_closepipes(_client);
				continue;
			}
            char* pChar;
            pChar = strtok(serial_str, "#");
            while (pChar != NULL)
            {
                printf ("%s\n", pChar);
                if (strstr(pChar, "ecid:") > 0)
                {
                    strncpy(serial_str, pChar, sizeof(serial_str));
                    break;
                }
                pChar = strtok(NULL, "#");
            }
            char* p = strchr(serial_str, '#');
            if (p) {
                *p = '\0';
            }
'''

After this modification idevicerestore fails at libidevicerestore at this point:
Sending iBEC (714456 bytes)...
ERROR: Unable to send iBEC component: Unable to upload data to device

taanders, 2018-08-31:
iTunes logs from older driver:
'''
[07:29:35.0567] iBoot build-version = iBoot-4076.70.15
[07:29:35.0569] iBoot build-style = RELEASE
[07:29:35.0569] unable to open device_map.txt: No such file or directory
[07:29:35.0570] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:35.0570] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:35.0570] requested restore behavior: Erase
[07:29:35.0575] requested restore behavior: Erase
[07:29:35.0576] unable to open device_map.txt: No such file or directory
[07:29:35.0576] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:35.0576] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:35.0578] AMDeviceIoControl: GetOverlappedResult failed
[07:29:35.0578] AMDeviceIoControl: pipe stall
[07:29:35.0578] USBControlTransfer: error 31, usbd status c0000004
[07:29:35.0578] command device request for 'getenv radio-error' failed: 2008
[07:29:35.0578] radio-error not set
[07:29:35.0579] unable to open device_map.txt: No such file or directory
[07:29:35.0579] <Recovery Mode Device 00000266A69C84E0>: production fused device
[07:29:35.0579] unable to open device_map.txt: No such file or directory
[07:29:35.0580] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:35.0580] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:35.0580] requested restore behavior: Erase
[07:29:35.0584] requested restore behavior: Erase
[07:29:35.0587] <Recovery Mode Device 00000266A69C84E0>: operation 4 progress -1
[07:29:35.0593] unable to open device_map.txt: No such file or directory
[07:29:35.0594] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:35.0594] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:35.0594] requested restore behavior: Erase
[07:29:35.0597] requested restore behavior: Erase
[07:29:35.0599] <Recovery Mode Device 00000266A69C84E0>: operation 43 progress -1
[07:29:35.0599] requested restore behavior: Erase
[07:29:35.0603] Max file size is: 268435456
[07:29:36.0612] No additional boot images found
[07:29:36.0612] <Recovery Mode Device 00000266A69C84E0>: operation 5 progress -1
[07:29:38.0637] unable to open device_map.txt: No such file or directory
[07:29:38.0637] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:38.0637] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:38.0653] AMDeviceIoControl: GetOverlappedResult failed
[07:29:38.0653] AMDeviceIoControl: pipe stall
[07:29:38.0653] USBControlTransfer: error 31, usbd status c0000004
[07:29:38.0653] command device request for 'getenv ramdisk-delay' failed: 2008
[07:29:40.0178] <Recovery Mode Device 00000266A69C84E0>: operation 6 progress -1
[07:29:40.0178] Max file size is: 268435456
[07:29:41.0194] <Recovery Mode Device 00000266A69C84E0>: operation 7 progress -1
[07:29:41.0196] Max file size is: 268435456
[07:29:41.0546] <Recovery Mode Device 00000266A69C84E0>: operation 8 progress -1
[07:29:41.0547] unable to open device_map.txt: No such file or directory
[07:29:41.0547] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[07:29:41.0547] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[07:29:41.0549] <Recovery Mode Device 00000266A69C84E0>: operation 9 progress -1
[07:29:41.0550] <Recovery Mode Device 00000266A69C84E0>: Recovery mode succeeded
[07:29:41.0550] Finished Recovery Restore Phase: Successful
[07:29:42.0600] Recovery mode device disconnected
'''

iTunes logs from 423.36.0.0 driver:
'''
[08:04:25.0250] iBoot build-version = iBoot-4076.70.15
[08:04:25.0252] iBoot build-style = RELEASE
[08:04:25.0253] unable to open device_map.txt: No such file or directory
[08:04:25.0254] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[08:04:25.0254] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[08:04:25.0254] requested restore behavior: Erase
[08:04:25.0259] requested restore behavior: Erase
[08:04:25.0265] unable to open device_map.txt: No such file or directory
[08:04:25.0268] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[08:04:25.0268] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[08:04:25.0270] AMDeviceIoControl: GetOverlappedResult failed
[08:04:25.0271] AMDeviceIoControl: failed, error 2, usbd status 00000000
[... 40 lines cut ...]
[08:04:34.0649] <Recovery Mode Device 0000021E6D91A040>: operation 8 progress -1
[08:04:34.0649] unable to open device_map.txt: No such file or directory
[08:04:34.0650] found device map entry for 0x00008960 0x00000002. boardConfig=n53ap platform=s5l8960x
[08:04:34.0650] _AMRestoreCopyDeviceMapPlistEntryForHardware: firmwareDirectory not in options
[08:04:34.0652] <Recovery Mode Device 0000021E6D91A040>: operation 9 progress -1
[08:04:34.0652] <Recovery Mode Device 0000021E6D91A040>: Recovery mode succeeded
[08:04:34.0652] Finished Recovery Restore Phase: Successful
[08:04:35.0715] Recovery mode device disconnected
'''

In libirecovery.c function irecv_usb_control_transfer() bRet gets false and last error code is 2:
'''
bRet = GetOverlappedResult(client->handle, &overlapped, &count, FALSE);
    if (!bRet)
    {
        printf("GetOverlappedResult %i\n", GetLastError());
    }
	CloseHandle(overlapped.hEvent);
	if (!bRet && bm_request_type != 65) {
        printf("\nDeviceIoControl no device found %i\n", bRet);
		CancelIo(client->handle);
		free(packet);
		return -1;
	}
'''

Johnny314, 2018-08-31:
@taanders  it seems issue with DeviceIoControl but 3utools and itunes is also using DeviceIoControl and able to restore the device with latest drivers as well...not sure might be possible they are using different dwIoControlCode
[image]

Johnny314, 2018-09-01:
Is it fixed? Actually i have fixed the issue but not sure fixed is valid or not can you send me your email?

nikias, 2018-09-01:
I don't understand your modifications. Why do you need to change the parsing? Also the line 669 you mention is this:
```
irecv_copy_nonce_with_tag(_client, "NONC", &_client->device_info.ap_nonce, &_client->device_info.ap_nonce_size);
```
The string between the `#` is this:

nikias, 2018-09-01:
Ah forget my comment, I didn't see the second sscanf of your changes since it was off screen, sorry.

nikias, 2018-09-01:
How about this patch:
https://gist.github.com/nikias/49d6bdcf75fb00207fb90dbe29dfafe2

nikias, 2018-09-02:
@Johnny314 if you have any fix it would be nice to post it here so others can profit from it too. Thanks.

taanders, 2018-09-03:
Nikias: your patch seems to work well with those new Apple drivers.

@Johnny314 Can you send working dwIoControlCode to this issue?

nikias, 2018-09-03:
@taanders pushed to master.

taanders, 2018-09-04:
Thanks @nikias for your commit .

I tried to set other DeviceIoControl dwIoControlCode values for those packets which bm_request_type is 65.

With value 0x2200B8 I got then error code -4 which means IRECV_E_INVALID_INPUT and with value 0x22000C I got error code -8 which means IRECV_E_USB_INTERFACE.

```
memset(&overlapped, 0, sizeof(overlapped));
overlapped.hEvent = CreateEvent(NULL, TRUE, FALSE, NULL);
if (bm_request_type == 65)
{
    bRet = DeviceIoControl(client->handle, 0x2200B8, packet, sizeof(usb_control_request) + w_length, packet, sizeof(usb_control_request) + w_length, NULL, &overlapped);
}
else
{
    bRet = DeviceIoControl(client->handle, 0x2200A0, packet, sizeof(usb_control_request) + w_length, packet, sizeof(usb_control_request) + w_length, NULL, &overlapped);
}

```

I'm not sure if those values are helpful for solving this issue.

Johnny314, 2018-09-06:
@nikias  @taanders  sorry guys for late reply @taanders just use this 0x2201b6 dwIoControlCode value and it will work.i have seen that in 3utools
ret = DeviceIoControl(client->handle, 0x2201b6, data, length, data, length, (PDWORD) transferred, NULL);
Thanks

taanders, 2018-09-07:
@Johnny314  Should there be also some other modifications on libirecovery? I was not able to finish device flashing by replacing only that line on IRECV_API int irecv_usb_bulk_transfer(...) to use 0x2201b6 dwIoControlCode. I tried to set that value 0x2201b6 also there but then device 

Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (714456 bytes)...
ERROR: Unable to send iBEC component: Unable to find device
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC

This code below seems to exit because bRet is set to false. It seems to happen on bm_request_type 65 package.
```
IRECV_API int irecv_usb_control_transfer(...)
...
	memset(&overlapped, 0, sizeof(overlapped));
	overlapped.hEvent = CreateEvent(NULL, TRUE, FALSE, NULL);
	DeviceIoControl(client->handle, 0x2200A0, packet, sizeof(usb_control_request) + w_length, packet, sizeof(usb_control_request) + w_length, NULL, &overlapped);
	WaitForSingleObject(overlapped.hEvent, timeout);
	bRet = GetOverlappedResult(client->handle, &overlapped, &count, FALSE);
	CloseHandle(overlapped.hEvent);
	if (!bRet) {
		CancelIo(client->handle);
		free(packet);
		return -1;
	}

	count -= sizeof(usb_control_request);
	if (count > 0) {
		if (bm_request_type >= 0x80) {
			memcpy(data, packet->data, count);
		}
	}
	free(packet);
...
```

Johnny314, 2018-09-07:
@taanders i just changed dwIoControlCode and able to restore device successfully

Orbif, 2018-09-07:
Hello,

Thank you for this patch, it works with new drivers in Windows 10.

Nevertheless, new drivers don't seems to be available in Windows 7, Windows Update indicate old drivers are up to date.

Is it possible to add compatibility with both old and new drivers ?

Thank you.

Johnny314, 2018-09-08:
@Orbif  i have tested iphone 7  in Windows 10 its working with both old and new driver
Thanks

airwolf2026, 2018-09-10:
```
if (endpoint==0x4) {
		ret = DeviceIoControl(client->handle, 0x2201B6, data, length, data, length, (PDWORD) transferred, NULL);
		if (ret == 0)
		{
			debug("irecv_usb_bulk_transfer DeviceIoControl2201B6 error: %d \n", GetLastError());
		}
		else
		{
			int lpBytesReturned = 0;
			unsigned char* InBuffer = 0;
			ret = DeviceIoControl(client->handle, 0x2200B8, &InBuffer, 4u, &InBuffer, 4u, &lpBytesReturned, NULL);
		}
	}
```
this code current work with both old and new driver

taanders, 2018-09-10:
Thanks @Johnny314, I got idevicerestore also working with adding that 0x2201B6 to IRECV_API int irecv_usb_bulk_transfer(...).

@nikias I think there is still some issue when parsing that device serial number line. With modifications on your earlier commit it seems to work ok with new Apple drivers but with older Apple drivers parsing gets confused to last line part "\0000".

I had problems when I tested with my old modifications from first commit. That code missed adding end line at the end of string. Those modifications caused then some strange actions later.

```
//Old driver result line
//\\?\usb#vid_05ac&pid_1281#cpid:8015_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000929960838402e_ibfl:3d_srnm:[c8pw28mfjc69]#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\0000
			
//New driver result line
//\\?\usb#vid_05ac&pid_1281&mi_01#8&7eb9421&3&0001#{b8085869-feb9-404b-8cb1-1e5c14fa8c54}\usb#vid_05ac&pid_1281#cpid:8015_cprv:11_cpfm:03_scep:01_bdid:0a_ecid:000929960838402e_ibfl:3d_srnm:[c8pw28mfjc69]#mux1

Lines below seem to work with old and new Apple drivers:
/* get DFU paths */
...
if (!p && ((sscanf(result, "\\\\?\\usb#vid_%*04x&pid_%*04x#%s#", serial_str) != 1) && (sscanf(p, "\\usb#vid_%*04x&pid_%*04x#%s", serial_str) != 1)) || (serial_str[0] == '\0')) {

/* get iBoot path */
...
if (!p || ((sscanf(result, "\\\\?\\usb#vid_%*04x&pid_%*04x#%s#", serial_str) != 1) && (sscanf(p, "\\usb#vid_%*04x&pid_%*04x#%s", serial_str) != 1)) || (serial_str[0] == '\0')) {
```

nikias, 2018-09-10:
@taanders hmm ok right... I didn't really test the code. I committed another fix for it.

taanders, 2018-09-11:
I think those dwIoControlCode values have been used long time. I found some other 7-year-old Ruby project where those dwIoControlCodes are listed. This dwIoControlCode change from 0x220195->0x2201B6 might have been some bug/security issue/protocol change on older Apple drivers which is now fixed on those new drivers.

https://github.com/lite/irestore-win32
https://github.com/lite/irestore-win32/blob/master/win32device.rb

0x22000C = reset
0x2200B8 = ack
0x2200A0 = control io
0x2201B6 = bulk transfer

nikias, 2019-01-28:
@airwolf2026 can you explain the second DeviceIoControl there? It seems to read a status code or something? It wasn't there before and it doesn't seem to be necessary for it to work...

nikias, 2019-01-28:
@taanders so you modified both irecv_usb_bulk_transfer and irecv_usb_control_transfer? Or just irecv_usb_bulk_transfer?

nikias, 2019-02-13:
Ok so I tried @airwolf2026's approach, but to make it work I had to change this part in `irecv_open_with_ecid`  to make it work:
```c
			error = irecv_usb_set_interface(client, 1, 0);
			/*error = irecv_usb_set_interface(client, 0, 0);
			if (client->mode > IRECV_K_RECOVERY_MODE_2) {
				if (client->mode > IRECV_K_RECOVERY_MODE_2) {
					error = irecv_usb_set_interface(client, 1, 1);
				}
			}*/
```
I took it from the win32device.rb - did you guys have to do that too @taanders @Johnny314 ?

airwolf2026, 2019-02-20:
> @airwolf2026 can you explain the second DeviceIoControl there? It seems to read a status code or something? It wasn't there before and it doesn't seem to be necessary for it to work...

By now ,I can't reproduce The second DeviceIoControl program flow. maybe reverse from another tool?

taanders, 2019-02-22:
Hi @nikias, sorry for late reply. I had that change only at irecv_usb_bulk_transfer function.

It was possible to bypass this error when that line changed: "ERROR: Unable to send iBEC component: Unable to upload data to device"

taanders, 2019-02-22:
I spent last week by testing situation like in this issue: https://github.com/libimobiledevice/idevicerestore/issues/132

I had some strange actions on Windows side. I had uninstalled those 423.36.0.0 drivers with pnputil /delete-driver oemXX.inf /force. Those drivers appeared still somewhere for devices which changed mode from normal mode -> recovery mode -> flashing mode.

I used USBDeview to get rid of 423 drivers.

It is strange that I was able to flash 4 ios devices succesfully at the same time with that Startech ST1030USBM USB hub sometimes multiple times sequentially.

When I disconnected devices and then restarted same test with same HW setup then it might work very different way. Sometimes only 2/4 devices were flashed successfully.

I got this error: ERROR: Unable to receive message from FDR 03CF0870 (-4). 0/2 bytes

Followed by endless entries of:
No data to read

It was strange that usually those failing flashed were coming from same USB ports of those hubs. 

Another strange thing is that I get maybe only 1/100 of those FDR errors if I use some other USB hubs like this cheap and awful (works great, just build quality is cheap): 
http://www.logilink.eu/Produkte_LogiLink/Notebook-Computerzubehoer/USB_Hubs/USB_20_Hub_10-Port_mit_Netzteil_UA0096.htm

Kanong017, 2019-03-23:
@taanders Is there any way to contact you?

yogendra123456, 2020-05-26:
Hi @taanders i  use 0x2201b6 dwIoControlCode. I tried to set that value 0x2201b6 also there but then device

Extracting iBEC.iphone6.RELEASE.im4p...
Personalizing IMG4 component iBEC...
Sending iBEC (714456 bytes)...
ERROR: Unable to send iBEC component: Unable to find device
ERROR: Unable to send iBEC to device.
ERROR: Unable to send iBEC 

Kindly provide me updated where is fail my code. or another solution for it...

Thanks in Advanced

yogendra123456, 2020-06-05:
hi @nikias and @taanders,
kindly give me some instruction about libmobiledevice for windows i found some error which is given below, what happen on window iphone is not placed in restore mode. while iTunes update the firmware at this stage and also 3uTools what happen with libmobiledevice code, it happen with usbmuxd or others class which is not connect device to properly to window system...
kindly provide me some update.......
Release.txt
thanks in advanced @nikias and @taanders,

yogendra123456, 2020-06-10:
Hi All,

i restore my device but it create error again and again on windows 10...

Extracting recoverymode@1334~iphone-lightning.im4p...
Personalizing IMG4 component RecoveryMode...
Extracting iBoot.n71.RELEASE.im4p...
Personalizing IMG4 component iBoot...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component RestoreSEP...
Extracting sep-firmware.n71.RELEASE.im4p...
Personalizing IMG4 component SEP...
Sending NORData now...
Done sending NORData
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Unmounting filesystems (29)
Checking filesystems (15)
ERROR: Unable to receive message from FDR 00000124E29C2130 (-7). 0/2 bytes

Details in attached file..
[ErrorRelease.txt](https://github.com/libimobiledevice/libirecovery/files/4757525/ErrorRelease.txt)

kindly provide the update what is the solution of this error, i am using latest libmobiledevice code with visual studio 2019...

thanks in advanced....