libimobiledevice/libirecovery #45: error while loading shared libraries: libirecovery.so.2: cannot open shared object file: No such file or directory (Linux Mint))

cutecodee, 2019-03-11:
tbianco@tbianco-pc:~/Scrivania/libirecovery-master$ irecovery
irecovery: error while loading shared libraries: libirecovery.so.2: cannot open shared object file: No such file or directory

shinvou, 2019-10-08:
Please run `echo $LD_LIBRARY_PATH` and post the output here.

BrandonPlank, 2020-01-23:
I have the same issue :/ here is my logs on Ubuntu 19.10 
root@brandonplank:~# irecovery
irecovery: error while loading shared libraries: libirecovery.so.2: cannot open shared object file: No such file or directory
root@brandonplank:~# echo $LD_LIBRARY_PATH

root@brandonplank:~# 

Yeah, nothing is shown with the command echo $LD_LIBRARY_PATH

BrandonPlank, 2020-01-23:
*UPDATE*

Run ldconfig in the cmd line and it will fix it :)