libimobiledevice/libirecovery #50: -n false positive; stays in DFU mode

cmdpwnd, 2019-07-20:
**Model:** `iPhone 5s`

```
root@kali:~# lsusb | grep Apple
Bus 004 Device 005: ID 05ac:1227 Apple, Inc. Mobile Device (DFU Mode)
root@kali:~# irecovery -vv -r
Attempting to connect... 
opening device 05ac:1227...
Setting to configuration 1
Setting to interface 0:0
Connected to iPhone6,1, model n51ap, cpid 0x8960, bdid 0x00
root@kali:~# irecovery -vv -m
Attempting to connect... 
opening device 05ac:1227...
Setting to configuration 1
Setting to interface 0:0
Connected to iPhone6,1, model n51ap, cpid 0x8960, bdid 0x00
DFU Mode
root@kali:~# irecovery -vv -n
Attempting to connect... 
opening device 05ac:1227...
Setting to configuration 1
Setting to interface 0:0
Connected to iPhone6,1, model n51ap, cpid 0x8960, bdid 0x00
Command completed successfully
root@kali:~# irecovery -vv -m
Attempting to connect... 
opening device 05ac:1227...
Setting to configuration 1
Setting to interface 0:0
Connected to iPhone6,1, model n51ap, cpid 0x8960, bdid 0x00
DFU Mode
root@kali:~# 
```
Side note: Is it possible to reflash & keep data intact?

nikias, 2019-07-20:
This is not a false positive. Also, `irecovery -n` is meant for Recovery, not DFU Mode. The device has broken boot images so it stays in DFU mode is my guess. Yes you can restore without losing device data using idevicerestore (if the user filesystem was not damaged).

cmdpwnd, 2019-07-21:
Alright, I thought irecovery was for recovery mode & DFU mode. You're saying use idevicerestore... and that it can upload the IPSW without wiping userdata while in DFU mode?

nikias, 2019-07-21:
If you just use `idevicerestore -l` or pass the ipsw directly as in `idevicerestore iPhoneXXX.ipsw` it will perform a "user upgrade install" which preserves user data under normal circumstances. Please mind though that I cannot guarantee for anything (e.g. if the user data partition is damaged it might decide to wipe it without warning).

nikias, 2019-07-21:
Also irecovery tool *is* for DFU mode, it's just that in DFU mode you can't do much except loading the next stage bootloader image.

cmdpwnd, 2019-07-21:
In passing the IPSW directly, is the decision to wipe the user partition controllable or is it in the guise of some internal continuity check on the device itself? It would be nice to be able to access & read/copy the user partition prior, but I've never seen any implementation of this documented.

cmdpwnd, 2019-07-25:
To summarize:
When in **DFU Mode**, `irecovery` *only* allows you to attempt rebooting the device into **Recovery Mode** via `irecovery -n`. If this fails or `irecovery -m` still shows `DFU mode` afterward, try using [`idevicerestore`](https://github.com/libimobiledevice/idevicerestore) to restore your IPSW (firmware). This will allow you to keep your user data (*almost always*).

closed 2019-07-25 by cmdpwnd