libimobiledevice/libirecovery #53: Shell mode doesn't really work

shinvou, 2019-10-08:
When starting the irecovery tool in shell mode, the output looks like so:

`shinvou@hjerne_en ~> irecovery -s`
`>`

or when entering commands (which actually execute)

`shinvou@hjerne_en ~> irecovery -s`
`> help`
`> `

I remember that some time ago, when starting in shell mode, irecovery would actually display some info like iBoot version and blah. Also, when executing commands I don't see a response from iBoot on the device. help should display a command list.

Do you have any clue?
I tried with different devices (iPhone 5s, iPhone X and iPad Air), both recovery and DFU mode and also on various versions of macOS (10.14, 10.15).

nikias, 2019-10-15:
Did you guys update to the latest codebase? I fixed a few things recently.

shinvou, 2019-10-15:
Latest codebase, yep.

nikias, 2019-10-15:
Works for me on macOS 10.14 - can you make sure it was compiled with IOKit and not libusb backend?

shinvou, 2019-10-15:
Definitely using IOKit.

nikias, 2019-10-15:
Can you run it with `-v` ?
This is how it looks like for me:
```
$ tools/irecovery -v -s
Attempting to connect...
Attempting to connect...
opening device 05ac:1281 @ 0x14200000...
Setting to configuration 1
Setting to interface 0:0
Connected to iPhone9,3, model d101ap, cpid 0x8010, bdid 0x0c
Setting to interface 1:1
Setting to interface 0:0

=======================================
::
:: iBoot for d10, Copyright 2007-2018, Apple Inc.
::
::	Local boot, Board 0xc (d101ap)/Rev 0xd
::
::	BUILD_TAG: iBoot-4513.270.14
::
::	BUILD_STYLE: RELEASE
::
::	USB_SERIAL_NUMBER: CPID:8010 CPRV:11 CPFM:03 SCEP:01 BDID:0C ECID:0011111111111111 IBFL:3D SRNM:[Dxxxxxxxxxxxxx]
::
=======================================

d64e4ac4ef51039:2123
d64e4ac4ef51039:2082
d64e4ac4ef51039:2093
d64e4ac4ef51039:2130
16fc6bd46eff0a3:277
3974bfd3d441da3:853
Entering recovery mode, starting command prompt
Setting to interface 1:1
Setting to interface 0:0
>
```

MrKris99, 2019-10-17:
Just compiled last version with IOKit, now it works fine for me.

shinvou, 2019-10-22:
I figured out what was wrong and it had nothing to do with libirecovery. Fixed the problem, works fine now.

closed 2019-10-22 by shinvou

asperq, 2020-11-19:
Sorry, im kind a a new in to this. 

Can someone please help me with compiling irecovery with IOKit files?

savindu-pasintha, 2024-10-11:
i also got issue shell with ipad 2