libimobiledevice/libirecovery #60: Problem during make

erdincbeldek, 2020-02-05:
Hi all
I'm trying to compile libirecovery on my arch linux but I didn't succeed on make phase. Is there anyone who can help about the error i got? Thank you.
[image]

nikias, 2020-02-05:
Hi, it built successfully. You only got warnings.

erdincbeldek, 2020-02-05:
Oh really? Thank you! But I still haven't achieved a working idevicerestore after all dependencies. Any thoughts?
[image]

sm-moshi, 2021-05-24:
The issue @erdincbeldek showed still exists. The following error comes up at idevicerestore compile time even with libirecovery installed.

configure: error: Package requirements (libirecovery-1.0 >= 1.0.0) were not met:

Package 'libirecovery-1.0', required by 'virtual:world', not found