libimobiledevice/libirecovery #61: support iPad8,9 and iPad8,10

DanTheMann15, 2020-03-19:
// iPad Pro 11-inch (2020) WiFi //
Model: iPad8,9
Board: j417ap
Platform: t8027
APBoardID: 0x3C

// iPad Pro 11-inch (2020) WiFi + Cellular //

Model: iPad8,10
Board: j418ap
Platform: t8027
APBoardID: 0x3E
BBChipID: 0x00000068

--------------------

IPSW Link: https://updates.cdn-apple.com/2020WinterFCS/fullrestores/041-42554/2971A61F-8071-43AC-93BB-F68D1BEAD94E/iPad_Spring_2020_13.4_17E255_Restore.ipsw

nikias, 2020-03-19:
Hmm isn't there also a 12.9 inch model?

DanTheMann15, 2020-03-20:
yes there is, but there is currently no info for those yet.

thats why i posted this for iPad8,9 and 8,10 since apple goofed-up and posted the wrong ipsw early.

nikias, 2020-03-20:
oh right... it was the fuckup haha

DanTheMann15, 2020-04-20:
Good mother of god apple, another device...
well, here are the specifications to help add support:

// iPhone SE (2020) //

Model: iPhone12,8
Board: d79ap
Platform: t8030
APBoardID: 0x10
BBChipID: 0x00000068

closed 2020-05-15 by nikias