libimobiledevice/libirecovery #62: Macos 10.15.4 PKG_CHECK_MODULES error during ./autogen.sh (and potensial fix)

stemnic, 2020-04-06:
Hi,

I tried compiling libirecovery today and ran into a error with `PKG_CHECK_MODULES`

```
checking for PTHREAD_PRIO_INHERIT... yes
./configure: line 17646: syntax error near unexpected token `libusb,'
./configure: line 17646: `			PKG_CHECK_MODULES(libusb, libusb-1.0 >= $LIBUSB_VERSION)'
```

Is this not the wrong syntax for PKG_CHECK_MODULES? [CMake](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html)

My fix was to change remove the comma in configure from `PKG_CHECK_MODULES`
`PKG_CHECK_MODULES(libusb libusb-1.0 >= $LIBUSB_VERSION)`

nikias, 2020-04-06:
It is autoconf, not cmake. And the syntax is definitely correct. Do you have `pkg-config` installed in the first place?

stemnic, 2020-04-06:
Yes pkg-config is installed though brew and all of the requirements mentioned in the README
```
stemnic@mac ~ % pkg-config --version
0.29.2
```
It compiled and installed just fine.

Though I see now that I don't get the error anymore when i started from scratch again with a new copy of the git repo. So i'm guessing I just encountered some weird behaviour, since I missed a couple of the dependencies when I tried to build the first time.

closed 2020-04-20 by stemnic

leehfather, 2021-03-05:
Same errors help me 

[image]

leehfather, 2021-03-05:
> Hi help me i have the same issue did you fix it ? 
> 
> 
> 
> I tried compiling libirecovery today and ran into a error with `PKG_CHECK_MODULES`
> 
> 
> 
> ```
> 
> checking for PTHREAD_PRIO_INHERIT... yes
> 
> ./configure: line 17646: syntax error near unexpected token `libusb,'
> 
> ./configure: line 17646: `			PKG_CHECK_MODULES(libusb, libusb-1.0 >= $LIBUSB_VERSION)'
> 
> ```
> 
> 
> 
> Is this not the wrong syntax for PKG_CHECK_MODULES? [CMake](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html)
> 
> 
> 
> My fix was to remove the comma from `PKG_CHECK_MODULES` in configure
> 
> `PKG_CHECK_MODULES(libusb libusb-1.0 >= $LIBUSB_VERSION)`
> 
> 
> 
>

nikias, 2021-03-05:
@leehfather that's not correct. You are missing pkgconfig on your system. If this is macOS, try this:
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls https://git.io/JLNVm
chmod +x ./limd-build-macos.sh
./limd-build-macos.sh
```