libimobiledevice/libirecovery #64: Example of rebooting to OR from recovery?

Adderally, 2020-06-01:
So basically my device is in **Recovery 0x1281** and i want to be able to exit recovery.
`irecv_init();`
`irecv_client_t client;`

`irecv_open_with_ecid(&client, (unsigned long long)0xMYECID);`
`irecv_reboot(client);`

at **`irecv_reboot(client);`** i'm getting an **Access Violation**. Not really sure what happens, cause sometimes the phone will reboot during this "break", but right back into recovery lol.

Any examples would be much love!
I

nikias, 2020-06-01:
Well the `irecovery` tool can do it:
https://github.com/libimobiledevice/libirecovery/blob/master/tools/irecovery.c#L543-L562

Can you run it in gdb/lldb and get a backtrace when it crashes?

Adderally, 2020-06-01:
The error i'm getting is
`Exception thrown at 0x78FC1B21 (irecovery.dll) in myprogram.exe: 0xC0000005: Access violation reading location 0xCCCCCD24.`

both `irecv_send_command(client, (const char*)"reboot")` and `irecv_reboot(client)` gives this.