libimobiledevice/libirecovery #73: Build fine, but "Segmentation fault (core dumped)" after irecovery --shell

Ofdas-23, 2020-11-06:
Hello, 
I finished trough the installation, but whenever I want to do something, it gives me the error "Segmentation fault (core dumped)". What can I do? If more info needed, I can provide, but right now I can think of any.

nikias, 2020-11-06:
You could try to run it in a debugger so we see what happens. Not sure what platform you are on but on linux for example you can use `gdb --args irecovery --shell`

Ofdas-23, 2020-11-06:
@nikias I dont know how  to use gdb... :/ what would the code look like? I use linux subsystem for windows, is that a problem?

Ofdas-23, 2020-11-06:
@nikias Well I dont know what that means or wether it is relevant or not, but I got this as output
irecovery.c: In function ‘print_usage’:
irecovery.c:378:25: error: expected ‘)’ before ‘PACKAGE_URL’
  378 |  printf("Homepage:    <" PACKAGE_URL ">\n");
---------------|-----------------^~~~~~~~~~~~
---------------|----------------- )
irecovery.c:379:25: error: expected ‘)’ before ‘PACKAGE_BUGREPORT’
  379 |  printf("Bug Reports: <" PACKAGE_BUGREPORT ">\n");
--------------|-----------------^~~~~~~~~~~~~~~~~~
--------------|-----------------)
irecovery.c: In function ‘main’:
irecovery.c:480:34: error: ‘PACKAGE_VERSION’ undeclared (first use in this function)
  480 |     printf("%s %s\n", TOOL_NAME, PACKAGE_VERSION);
---------------------|--------------------^~~~~~~~~~~~~~~
irecovery.c:480:34: note: each undeclared identifier is reported only once for each function it appears in

nikias, 2020-11-06:
I think the subsystem wouldn't work with USB devices, but I can't say for sure.