libimobiledevice/libirecovery #79: Help me it gives me 2 error

leehfather, 2021-03-05:
[image]

PKG_CHECK_MODULES (libusb, libusb-1.0 > $LIBUSB_VERSION)'
./configure: line 17687

I did all the readme it gives this error at ./autogen.sh

nikias, 2021-03-05:
```
mkdir -p limd-build
cd limd-build
curl -o ./limd-build-macos.sh -Ls https://git.io/JLNVm
chmod +x ./limd-build-macos.sh
./limd-build-macos.sh
```

leehfather, 2021-03-05:
I am on windows 10 using virtual box ( kali linux ) i will try what you send to me and tell you what happened thanks for your replay mate .

leehfather, 2021-03-05:
No it didn't work its for macos i am using linux

nikias, 2021-03-05:
See here: https://github.com/libimobiledevice/libimobiledevice/issues/1067#issuecomment-726234592