libimobiledevice/libirecovery #105: iPad Mini 2 issue

lorenzoferron98, 2022-09-19:
I tried to send iBEC to my [iPad Mini 2](https://ipsw.me/iPad4,4/info). But at first attempt `irecovery`failed
```bash
% irecovery -v -f sshramdisk/iBEC.img4  
Attempting to connect... 
opening device 05ac:1227 @ 0x14200000...
Setting to configuration 1
Setting to interface 0:0
Connected to iPad4,4, model j85ap, cpid 0x8960, bdid 0x0a
DFU ERROR, issuing CLRSTATUS
Unable to upload data to device
```
At second attempt it failed with progress bar
```bash
% irecovery -v -f sshramdisk/iBEC.img4
Attempting to connect... 
opening device 05ac:1227 @ 0x14200000...
Setting to configuration 1
Setting to interface 0:0
Connected to iPad4,4, model j85ap, cpid 0x8960, bdid 0x0a
[====================================              ] 70.2%Unable to upload data to device
```
I compiled tool on my macOS 12.6 without problems. I successfully sent iBSS but no iBEC. I have no problems with my iPhone X.

nikias, 2022-09-20:
Can you check if it works when you change this line:
https://github.com/libimobiledevice/libirecovery/blob/master/src/libirecovery.c#L2719
from
```c
	int packet_size = recovery_mode ? 0x8000 : 0x800;
```
to
```c
	int packet_size = recovery_mode ? 0x2000 : 0x800;
```
?

nikias, 2022-09-20:
Oh hmm just saw it fails in DFU mode already, so this won't help...

lorenzoferron98, 2022-09-20:
Thank u for the answer. I tried to edit source but nothing has changed. I noticed that after send iBSS on my **iPad Pro** backlight is present but if I use iPad Mini 2 not.
`irecovery` doesn't disconnect and reconnect iDevice**s**. Anyway with iPad Mini 2 I tried to disconnect and reconnect manually before to load iBEC unsuccessfully.