libimobiledevice/libirecovery #107: I'm Getting This Error Every Time I Run iRecovery

XboxOneSogie720, 2022-12-23:
I just finished building libirecovery on Linux Mint. Everything went fine, and it installed correctly. I can even press tab while typing the command in terminal, meaning that in theory, irecovery should now work. However, this is not the case. When running any command, it gives me this error, shown in the attachment.
[image]

nikias, 2022-12-23:
Have you tried to run `sudo ldconfig` ?

XboxOneSogie720, 2022-12-23:
@nikias I didn't even know that command existed. It'd be nice if it was in the readme, and if it wasn't, then I just missed it. Anyways, that worked. Thanks!

closed 2022-12-23 by XboxOneSogie720