libimobiledevice/libirecovery #125: Failed to set interface, error -8

mast3rz3ro, 2023-12-09:
Hi,

I have successfully compiled the libirecovery with mingw64 on Msys2 by cloning it with git

Now when i try to use it i got this output:

```
# irecovery -q -v
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
Attempting to connect... 
Setting to interface 0:0
Failed to set interface, error -8
ERROR: Unable to set device interface
# irecovery -V
irecovery 1.1.0-9-g479b1b1
```

nikias, 2023-12-09:
What device and in which mode?

mast3rz3ro, 2023-12-10:
I have tested this on iPhone 11, and iPhone 7 Plus in recovery mode

using the [L1ghtmann](https://github.com/L1ghtmann/libimobiledevice/releases/) release kinda presents same error but it's works.

nikias, 2023-12-10:
I just pushed another commit, maybe that will solve the issue.

mast3rz3ro, 2023-12-10:
Awesome! Your commit has fixed the error completely!

Now the output are like this:
```
irecovery -q -v
Attempting to connect...
Setting to interface 0:0
Connected to iPhone12,1, model n104ap, cpid 0x8030, bdid 0x04
CPID: 0x8030
CPRV: 0x11
BDID: 0x04
ECID: 0x000408c11afb802e
CPFM: 0x03
SCEP: 0x01
IBFL: 0x3d
SRTG: N/A
SRNM: F4G*******
IMEI: N/A
NONC: f500fdf3e7eab20922637951e2948453ecd1dbb7163ec5acd4333c236ba4066c
SNON: f0ed745594736e36ea86406d0c54082981db29c9
MODE: Recovery
PRODUCT: iPhone12,1
MODEL: n104ap
NAME: iPhone 11
```

The L1ghtman release was returning this output:
```
irecovery -q -v
Attempting to connect...
Setting to interface 0:0
WARNING: set interface failed, error -8
Connected to iPhone12,1, model n104ap, cpid 0x8030, bdid 0x04
CPID: 0x8030
CPRV: 0x11
BDID: 0x04
ECID: 0x000408c11afb802e
CPFM: 0x03
SCEP: 0x01
IBFL: 0x3d
SRTG: N/A
SRNM: F4G*******
IMEI: N/A
NONC: f500fdf3e7eab20922637951e2948453ecd1dbb7163ec5acd4333c236ba4066c
SNON: f0ed745594736e36ea86406d0c54082981db29c9
MODE: Recovery
PRODUCT: iPhone12,1
MODEL: n104ap
NAME: iPhone 11
```

Thank you for your efforts :-)

nikias, 2023-12-10:
Aweseome, thanks for confirming.

closed 2023-12-10 by nikias